RealNobser/BetterDuinoFirmwareV4 #1: Servo Adjustments not Working

darkoperator, 2024-02-01:
I'm trying to adjust the servo travel to be 45 degrees, but it keeps doing its full travel movement.

 ```
#SO010045
Setup command
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒
OK
:OP01
OK
:CL01
OK
```

I event tried compiling the firmware but once uploaded setting the open value to 1500 it will not take commands via serial

darkoperator, 2024-02-01:
Compilation of firmware shows not errorr. Once flashed I get the marcduino version number in the terminal and that it it. It will not take commands 

```
 *  Executing task: C:\Users\Carlos Perez\.platformio\penv\Scripts\platformio.exe run --environment ATmega328P 

Processing ATmega328P (platform: atmelavr; board: ATmega328P; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega328P.html
PLATFORM: Atmel AVR (5.0.0) > ATmega328P/PA
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 32KB Flash
DEBUG: Current (avr-stub) External (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr-minicore @ 3.0.0
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SendOnlySoftwareSerial @ 1.0.0+sha.a1d3d44
|-- VarSpeedServo @ 0.0.0+20240201151742.sha.bb99aa1
|-- Wire @ 1.1
|-- EEPROM @ 2.0
Building in release mode
Compiling .pio\build\ATmega328P\src\Holo.cpp.o
Compiling .pio\build\ATmega328P\src\MDuinoBase.cpp.o
Compiling .pio\build\ATmega328P\src\MDuinoBodyMaster.cpp.o
Compiling .pio\build\ATmega328P\src\MDuinoDome.cpp.o
Compiling .pio\build\ATmega328P\src\MDuinoDomeMaster.cpp.o
Compiling .pio\build\ATmega328P\src\MDuinoDomeMasterChopper.cpp.o
Compiling .pio\build\ATmega328P\src\MDuinoDomeSlave.cpp.o
Compiling .pio\build\ATmega328P\src\MDuinoSequencer.cpp.o
Compiling .pio\build\ATmega328P\src\MDuinoSound.cpp.o
Compiling .pio\build\ATmega328P\src\MDuinoStorage.cpp.o
Compiling .pio\build\ATmega328P\src\Panel.cpp.o
Compiling .pio\build\ATmega328P\src\main.cpp.o
Compiling .pio\build\ATmega328P\lib8db\SendOnlySoftwareSerial\SendOnlySoftwareSerial.cpp.o
Compiling .pio\build\ATmega328P\libfb9\VarSpeedServo\VarSpeedServo.cpp.o
Compiling .pio\build\ATmega328P\lib75e\Wire\Wire.cpp.o
Compiling .pio\build\ATmega328P\lib75e\Wire\utility\twi.c.o
Archiving .pio\build\ATmega328P\libFrameworkArduinoVariant.a
Compiling .pio\build\ATmega328P\FrameworkArduino\CDC.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\HardwareSerial0.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\HardwareSerial1.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\HardwareSerial2.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\HardwareSerial3.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\PluggableUSB.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\Print.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\USBCore.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\WInterrupts.c.o
Compiling .pio\build\ATmega328P\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\WString.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\abi.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\hooks.c.o
Compiling .pio\build\ATmega328P\FrameworkArduino\main.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\new.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\wiring.c.o
Archiving .pio\build\ATmega328P\lib8db\libSendOnlySoftwareSerial.a
Compiling .pio\build\ATmega328P\FrameworkArduino\wiring_analog.c.o
Compiling .pio\build\ATmega328P\FrameworkArduino\wiring_digital.c.o
Compiling .pio\build\ATmega328P\FrameworkArduino\wiring_extras.cpp.o
Compiling .pio\build\ATmega328P\FrameworkArduino\wiring_pulse.S.o
Compiling .pio\build\ATmega328P\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\ATmega328P\FrameworkArduino\wiring_shift.c.o
Archiving .pio\build\ATmega328P\lib75e\libWire.a
Archiving .pio\build\ATmega328P\libfb9\libVarSpeedServo.a
Archiving .pio\build\ATmega328P\libFrameworkArduino.a
Linking .pio\build\ATmega328P\firmware.elf
Checking size .pio\build\ATmega328P\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [======    ]  60.6% (used 1242 bytes from 2048 bytes)
Flash: [==========]  96.6% (used 31646 bytes from 32768 bytes)
Building .pio\build\ATmega328P\firmware.hex
====================================================================================================== [SUCCESS] Took 1.89 seconds ======================================================================================================
 *  Terminal will be reused by tasks, press any key to close it. 
 ```

RealNobser, 2024-02-01:
Hi!

Some questions:
a) which terminal program do you use to send the MarcDuino commands?
b) make sure to terminate the commands by CR only, not CRLF. Configure your terminal to just send CR when pressing enter
c) degrees and values are servo related, not panel related. So a value of 45 deg means, the servo moves to 45 deg position. It does not mean, that your panel is opened half way up.

Nevertheless I will check your issue and report back.

darkoperator, 2024-02-01:
I'm using Putty. don't think it is the terminal app given that the release hex once installed works, works on the marduino v3 and it only fails on my compiled version. How can I adjust it so the panel only opens at a given position?

darkoperator, 2024-02-01:
the tasks I need to perform are:
* Invert direction for some panels
* Change the amount of degrees or position they open to 

Sadly I can figure it out from the documentation since  I tried #SWxx and it opens and closes in the same directions

RealNobser, 2024-02-01:
Did you take a look here? https://github.com/RealNobser/BetterDuinoFirmwareV4?tab=readme-ov-file#best-practice-panel-adjustment

darkoperator, 2024-02-01:
yep but move commands come back as invalid

:MV010000
**Invalid Panel Command
:MV010180
**Invalid Panel Command

darkoperator, 2024-02-01:
I even copy pasted the velues from the documentation and tested an open and close to ensure it is not the terminal and it opens and closes with those commands, mut MV commands return as invalid
```
:MV010900
**Invalid Panel Command
:MV011750
**Invalid Panel Command
:OP01
OK
:CL01
OK
```

RealNobser, 2024-02-01:
Which Release do you use? The releases published here have no feedback messages activated. Normally debug_msg is OFF due to memory! Tried on my desk, it is all fine. So did you flash V1.0.3?

RealNobser, 2024-02-01:
Please post your config.h if you made any changes and compiled an own release. Otherwise I cannot check.

darkoperator, 2024-02-01:
using release hex for BetterDuinoV103.hex

[image]

darkoperator, 2024-02-01:
WhenI tried building my own 

```
#ifndef __CONFIG_H__
#define __CONFIG_H__

// Specific Astromech, Choose one
#define R2D2
// #define CHOPPER

// #define DEBUG_MSG                    // Additional Log Messages to Serial
#define INCLUDE_BODY_MASTER             // Uncomment to include Body Controller in combined firmware
// #define INCLUDE_CLASSIC_I2C_SUPPORT  // Uncomment to include classic Seriial to I2C support.
#define INCLUDE_I2C_SLAVE               // Uncommend for new I2C slave mode

#define DEDICATED_FIRMWARE   // Separate Hex Files for Master/Slave/Body
// Choose one:
 #define DEDICATED_MASTER
// #define DEDICATED_SLAVE
// #define DEDICATED_BODY

#define SERIAL_BAUD         9600
#define SERIAL_SLAVE_BAUD   9600
#define SERIAL_MP3_BAUD     9600
#define SERIAL_TEECES_BAUD  2400
#define SERIAL_MAGIC_BAUD   9600

#define SERIALBUFFERSIZE    16

#define HEARTBEAT_MILLIS    1000

#define MINRANDOMPAUSE 6        // In Seconds!
#define MAXRANDOMPAUSE 12       // In Seconds!

#define MAX_SOUND_BANK  9
#define MAX_BANK_SOUND  25

// Original MarcDuino Default Values
#define PANEL_OPN   1500 // Open
#define PANEL_CLS   2000 // Closed

#define HOLO_MIN    1000
#define HOLO_MAX    2000

#define SERVO_SPEED_FULL        0
#define SERVO_SPEED_FAST        255
#define SERVO_SPEED_MEDIUM      175
#define SERVO_SPEED_SLOW        100
#define SERVO_SPEED_SUPER_SLOW  50

#define SERVO_BUZZ_MILLIS       2000

#define COMMAND_DELAY           50

// #define DIGITAL_JEDI

// DO NOT EDIT BELOW THIS LINE UNLESS YOU REALLY KNOW WHAT YOU ARE DOING
// ------------------------------------------------

//
// Module Specific stuff
//
#define VERSION         "4.1.4"
#define HW_VERSION      "1.5.3"
#define CONFIG_VERSION  41

#define MAX_MARCUDINOSERVOS 13
#define MAX_SEQUENCE_COMPLETION_CALLBACKS 5
#define MAX_MARCDUINOHOLOS  3

// Panel Sequence Position. No PWM value any more! Don't touch here!
// Will be opening level in percent in the future
#define _NP     255
#define _CLS    0
#define _OPN    100

// Generic GPIOs
#define P_I2C_SCL   19
#define P_I2C_SDA   18
#define P_LED2      17
#define P_RC_IN      8
#define P_SERIAL1   14
#define P_SERIAL2   15
#define P_AUX1      16

// Dome Master
#define P_SERVO_01  2
#define P_SERVO_02  3
#define P_SERVO_03  4
#define P_SERVO_04  5
#define P_SERVO_05  6
#define P_SERVO_06  9
#define P_SERVO_07  10
#define P_SERVO_08  11
#define P_SERVO_09  12
#define P_SERVO_10  13
#define P_SERVO_11  7
#define I2C_DOME_MASTER 50

// Dome Slave
#define P_HPF_V     2
#define P_HPF_H     3
#define P_HPR_V     4
#define P_HPR_H     5
#define P_HPT_V     6
#define P_HPT_H     9
#define P_FL        10
#define P_RL        11
#define P_TL        12
#define P_SERVO_12  13
#define P_SERVO_13  7
#define I2C_DOME_SLAVE  51

// Body Master
#define P_DPL            2
#define P_UTIL_ARM_U     3 
#define P_UTIL_ARM_L     4
#define P_LEFT_BDY_DR    5
#define P_LEFT_ARM       6
#define P_LEFT_ARM_TOOL  9
#define P_RIGHT_BDY_DR   10
#define P_RIGHT_ARM      11
#define P_RIGHT_ARM_TOOL 12
#define P_CHRG_BAY_DR    13
#define P_RFU            7
#define I2C_BODY_MASTER 52

#endif // __CONFIG_H__
```

darkoperator, 2024-02-01:
JIC boards are 1.5 Rev 3

RealNobser, 2024-02-01:
I cannot reproduce. What I did:

- Using a fresh Windows Laptop
- Download BetterDuinoV103.hex
- Flash BetterDuinoV103.hex using myAVRProgTool (https://shop.myavr.de/index.php?sp=download.sp.php&suchwort=DL112)
- Connecting serial interface, 9600 baud, 8N1.
- Using Putty in default config to connect to com port
- :MV010900 works as well as :MV011750 or :MV010090 ...
- V1.0.3 **does not echo commands and does not send feedbacks**, so "**Invalid Panel Command" ist NOT from V1.0.3

It looks like you are still using one of the beta versions.

What do you get when sending #RSET?

darkoperator, 2024-02-01:
Not in front of the computer at the moment but mow realize I have a marcduino v3 slave connected to the betterduinov4  and that might be the one returning those messages

RealNobser, 2024-02-01:
I checked the original MarcDuino V3 sources.

#if _ERROR_MSG_ == 1
const char strPanelCmdErr[] PROGMEM="**Invalid Panel Command\r\n";
#endif

Your putty is connected to the board with the original V3 firmware and not to the board with BetterDuino. So the BetterDuino commands will not work for sure.

Closing this issue.

closed 2024-02-01 by RealNobser

RealNobser, 2024-02-01:
Which is fine. BetterDuino Masters are working fine with MarcDuino Slaves. But your terminal is not connected to BetterDuino Master. You are sending BetterDuino commands to the MarcDuinoV3 slave. That does not work. Neither with BetterDuino nor with MarcDuino.

darkoperator, 2024-02-02:
just got back home .. the slave is not connected to the master

darkoperator, 2024-02-02:
I think I figured the issue with my 3 boards. the eFuse us not set correctly. I see it in the verbose programer output that it failed to set it.  I programed the board outside the dome, nothing but serial an programer connected and I see it was not resetting after programing. I could tell it, it is a body marcduino but when I do the movement command it fails with the same message as before. 

```
MarcDuino HP Control v3.4
Enter HP or Display command starting with '*' or '@'
Calc CRC: 7 StoredCRC: FF00
EEPROM Corrupt or never written.  Writing defaults.
Generating new CRC.
Calc CRC: 7 StoredCRC: FF00
New Defaults and CRC Written to EEPROM

OK
#RSET
Setup command
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒

OK

OK
#MD02
Setup command
▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒

OK

OK
:MV010900
**Invalid Panel Command

```

Guess I will need to buy new chips

RealNobser, 2024-02-02:
Once again: The log you are sending is MarcDuino V3, NOT BetterDuino. "MarcDuino HP Control V3.4" is NOT my firmware. You cannot send #RSET or :MV commands to V3.4. Check your wiring or whatever.