RealNobser/BetterDuinoFirmwareV4 #2: Program size to big?

PatrickSteiner, 2024-04-03:
Hi, followed your instructions on building and uploading BetterDuino via VisualStudioCode, but am getting 

`Error: The program size (32372 bytes) is greater than maximum allowed (32256 bytes)`

Is there more to change in platformio.ini than the USB Port?

RealNobser, 2024-04-03:
Hi!

I fixed the fuse settings, thanks for the hint!

Regarding the file size:
In the PlatformIO extension, go to "Dependencies/Update". This will install an updated ATmega328P Dependency, which is smaller than the old ones. It is still tight, then, but the standard builds should work again. 

Please give me an hint if that worked for you!

PatrickSteiner, 2024-04-04:
Hi,

compile now works. I only struggle with the upload in VisualStudio, works with ArduinoIDE but not with PlatformIO. Have changed the port but still get `avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00`

RealNobser, 2024-04-04:
Which progger do you use?

PatrickSteiner, 2024-04-04:
I did various experiments. I did try to flash an Arduino Pro Mini via this programmer: https://www.reichelt.de/entwicklerboards-usb-ttl-schnittstellenwandler-ft232-debo2-usb-ttl-p282731.html?PROVID=2788

As this didn't work, I tried generally flash an Arduino Nano directly vi USB Cable, which worked with the Blink Example via ArduinoIDE but not via PlatformIO ... I guess that there is some misunderstanding or wrong config on my side :-(

Any idea what I might be doing wrong?

PatrickSteiner, 2024-04-04:
Not sure if I'm on the right path, but changing `board = ATmega328P` to `board = nanoatmega328new` in platform.io allowed my test to flash and work. Unfortunately changing this does create compile errors in BetterDuino, as `printf`is not a member function of `SendOnlySoftwareSerial` or any of its inherited classes.

Am I onto something or just down a rabbit hole?

RealNobser, 2024-04-04:
MarcDuino boards are designed to be programmed via ISP, not via serial bootloader. There is not enough memory for bootloader AND MarcDuino/BetterDuino. So do you have a Nano board with ISP-Header? (e.g. https://www.roboter-bausatz.de/p/nano-mega328p-v3.0-usb-ch340g-modul). ISP is the 6 pin block on the rear.

PatrickSteiner, 2024-04-04:
Makes sense ... I do have the ISP Header soldered to my Nano. I have also tried to flash my Pro Mini via my programmer but ran into the same issue. If I use `board = ATmega328P` if will not flash, but it will with `board = nanoatmega328new`. 
This is also re-producible with ArduinoIDE. Using processor `ATmega328P` get's timeouts but using `ATmega328P ( Old Bootloader )` works.

Could this be caused by a non-compatible programmer ( see link to mine above )? Happy to buy the USB ISP-Programmer für Atmel AVR from Diamex Shop, as linked in your instructions, if that solves my issues.

RealNobser, 2024-04-04:
I will check your programmer in detail, later. The Diamex works fine with the configuration in the repository. It works with original MarcDuino boards as well as with Arduino Nanos or Pro Minis (when using the ISP-Headers on the corresponding AIO board).

PatrickSteiner, 2024-04-04:
Error was, as expected, on my side. My Programmer is FTDI - will order an ISP capable one.

Thanks for your patience with me.

closed 2024-04-04 by RealNobser