RealNobser/BetterDuinoFirmwareV4 #6: 6 servo imputs for lower dome.

stewartgill1977, 2024-04-14:
6 servo inputs for lower dome but new dome v4 has 7 servos is there a extra input for it. I know there is one for the magic panel but not sure where the 7 lower panel one is. 
[image]

RealNobser, 2024-04-14:
Mr. Baddeley added two additional panels, the small ones right and left beside the logic light. So there a 8 lower panels and 4 pie panels now. These are not supported by default BetterDuino or MarcDuino boards or animations. Of course you can use Servo 11 on Master board and Servo 12 and 13 on Slave board (aka 10 and 11) to control them nevertheless. But then you are on your own for adapting the animations etc. for now.

stewartgill1977, 2024-04-14:
Hi no there are 7 one extra hmmm ill take a look at the code and see if i can work it out. Im not much of a programmer but can do a little thanks
[image]

stewartgill1977, 2024-04-14:
I think i get it i just edit this eg the servos like 12 or 13 to opn the servo i use ?
[image]

RealNobser, 2024-04-18:
Overwrite might happen. So the best way would be, you create a fork for yourself and only sync changes from my side to your repository if needed.

closed 2024-04-18 by RealNobser