RealNobser/BetterDuinoFirmwareV4 #7: Holo lights

stewartgill1977, 2024-04-18:
Hi just finished wiring everything up and none of the holos light up on start up or commands. I tested them before i put them in on the desk and the worked. I have a funny felling it may of been the older ver of the firmware but i can not remember. I have checked voltage and im getting 5v to the lights. I also plugged then into a servo tester and they also did not turn on im not shore if they need a signal or something and that's why they do not turn on with the servo tester ??. Sorry dont know much about them but this is what im using. https://electropeak.com/7-bit-ws2812-round-led-light-module WS2812 7-Bit RGB LED Ring Module

RealNobser, 2024-04-18:
Which version of Betterduino? Have you configured the NeoPixels correctly (correct number of LEDs)? Did you use the NeoPixel firmware or the standard firmware? NeoPixels need some memory, so you will find a master+slave-only firmware with NeoPixel support separately in each version.
Here is the recent officially supported version: https://github.com/RealNobser/BetterDuinoFirmwareV4/releases/tag/v1.0.11

stewartgill1977, 2024-04-18:
yep thats the ver im running the latest.  "Have you configured the NeoPixels correctly" I just plugged them in was working before but not shore what firmware i was using may of been the older but im on the newest one now have not entered any commands do i need to to get them to work with the latest ver?

stewartgill1977, 2024-04-18:
ooo do i need the bottom ver??? that could be why i have loaded the top ver?
[image]

RealNobser, 2024-04-19:
Yes...

closed 2024-04-19 by RealNobser