RealNobser/BetterDuinoFirmwareV4 #8: Use #SWxx for Body Servos

PatrickSteiner, 2024-04-18:
Hi,

the instructions state that Panel directions ( previously 'reverse' ) can be changed via #SWxx with xx being the Panel. I did that without any issues for the Done Panels. What about the Servos connected via Body Master on Astrocomm Boards? Do I use the same numbering 01 - 10 like when connecting to the Dome Master?

PatrickSteiner, 2024-04-20:
OK. Was brave and used #SW command also to Servos for Body ... works like for the Panels, also numbering the servos based on where they are connected to Body Master, numbering from 1 to 10.

closed 2024-04-20 by PatrickSteiner

RealNobser, 2024-04-20:
Hi Patrick!

Yes, the Master-Boards all work the same way. You can adjust or switch the servos at will. Just the slave board has a different assignment of the servo ports 7-9.