RealNobser/BetterDuinoFirmwareV4 #9: betterduino sound commands

stewartgill1977, 2024-06-09:
Hi hope you can help i have noticed after changing to betterduino some sounds are not triggering with the sequence trigged eg the beep cantina the panels work but the sound for that seq dose not trigger anymore as showing in the picture the command im using. I have not changed anything apart from the firmware is there a bug or do i need to change something?. I have notices some buttons now have different  seq after the flash but that's no problem just trying to get the sound working. In saying that the leia one is working and scream one is fine as well so im not two shore what going on. Also CONFIGURE: Arrow Down + CIRCLE i had set to disco witch dose the panels but not the sounds at well 

[Untitled](https://github.com/RealNobser/BetterDuinoFirmwareV4/assets/37970956/483c755b-0086-4848-b62d-06a3930bd5e3)
[image]

stewartgill1977, 2024-06-09:
just tired to assign the command disco to another button and yep get the panels but no sound :SE09 very strange as it was find with the old firmware hmmm

RealNobser, 2024-06-09:
Hi Steward. First of all: Don't mix things up. MarcDuino/BetterDuino has nothing to do with your code snipplet above. The one above is ShadowMD. BetterDuino has nothing to do with button assignment.

**HINT**
BetterDuino supports multiple sound interfaces. Have a look here: https://github.com/RealNobser/BetterDuinoFirmwareV4?tab=readme-ov-file#setup-commands

Configure the right one, default is SparkFun MP3

#MPxx | Set MP3-Player Type· 

#MP00 : SparkFun MP3 Trigger· 
#MP01 : DFPlayer· 
#MP02 : Vocalizer

Please read the documentation an configure BetterDuino matching your hardware setup.

stewartgill1977, 2024-06-09:
HI im a little confused lol sorry. I have set it all up using the commands on your page yes im using mp3 trigger. I have entered :SE09 (disco) and :SE07 cantina dance i get the panels but no sound trigger for those two commands all others are working. do those two now have no sound attached to them now? 

thanks :)

RealNobser, 2024-06-10:
They have, they work fine here. Can you please send a list of the directory structure of you MP3 SD-card directory?

stewartgill1977, 2024-06-10:
[image]

stewartgill1977, 2024-06-10:
its not a Biggy its only two sequences that do not trigger the sound but i found a work around as in the pic. All others work as intended 
[image]

RealNobser, 2024-06-10:
$86 means "bank 8, song 6"

There had been an change in the most recent MarcDuino-Versions regarding the numbers of the sound banks and songs. BetterDuino also uses that mapping. Use the following numbering:

201 Beep Cantina
202 Star Wars
203 Imperial
204 Mana Mana
205 Cantina
206 Disco
207 R2 Star Wars
208 R2 Imperial
255 Startup Sound

stewartgill1977, 2024-06-10:
arrr ok so this is why maybe its not trigging the sound? Do you have a link to the updated sounds list or is it just these ones that have changed? and all i have to do is renumber the files i have to the numbers you show here?

RealNobser, 2024-06-10:
It is just the music files. They moved from 176++ to 201++

stewartgill1977, 2024-06-10:
great thanks for the info )

stewartgill1977, 2024-06-10:
so should look like this then just copy to sdcard like that and all should be good ? 
[image]

RealNobser, 2024-06-10:
Should work that way. Give it a try...

stewartgill1977, 2024-06-20:
hi again im having a little trouble understanding the sound files and triggering them. it seems that using the number of the track dose not play it or plays the wrong one. Would you know how to trigger a sound file like disco witch is (206 Disco) i tried putting command Serial1.print("$206\r"); but its not working. I know im just doing something wrong because it just plays the wrong track i just need to know how the order works thanks

stewartgill1977, 2024-06-20:
hello it looks like the play sound command $x will not play the bank at random eg i use  Serial1.print("$9\r"); to play music in bank 9 at random but nothing happens also tired other banks. It does how ever work if you also add the track number eg Serial1.print("$921\r"); is there a bug im trying to assign a button to play all music in the bank. 
[image]

RealNobser, 2024-06-27:
If this one is part of the MarcDuino standard, I will implement it. Will reopen that issue and check soon.

RealNobser, 2024-06-28:
Official MarcDuino code:

	/////////////COMMAND VOCABULARY///////////
	// Play sound command by bank/sound numbers
	// $xyy
	// x=bank number
	// yy=sound number. If none, next sound is played in the bank
	//

So $x ist allowed, but does not play random, but next sound. Nevertheless Betterduino does not support that, yet.

stewartgill1977, 2024-06-28:
I got  it by using this code so now plays random 

[image]

RealNobser, 2025-05-13:
Fixed with Release V1.4.4: https://github.com/RealNobser/BetterDuinoFirmwareV4/releases/tag/v1.4.4