RealNobser/BetterDuinoFirmwareV4 #10: Benduino v1.5, no serial response

neromir, 2024-06-15:
Hey, I'm just getting started with trying to get Betterduino put onto my Benduino AIO v1.5. I'm using the [Pololu USB AVR programmer v2.1](https://www.pololu.com/docs/0J67) and am having some issues getting a response back from the board after programming.

So far, programming it with platformio via CLion has been fine-- I checked out the v1.0.11 tag. But I haven't been able to get a serial response from the newly programmed arduino at all over the programmer's TTL connection. (I still have it plugged into the ISP header on the AIO). I've verified that the programmer itself works for TTL by setting it up in loopback mode (just connecting its TX and RX pins). 

With the stock marcduino firmware it came with I was able to use an ESP8266 to communicate with the AIO board with the R2Touch app. 

Any ideas? I'm hoping there's something obvious I'm doing wrong.

RealNobser, 2024-06-16:
I don't know the Pololu in details, but the programmers I normally use (Diamex S2) can either be in ISP mode or in Serial mode, not both in parallel.

I'm wondering how you connect TX/RX to the AIO. The AIO does not "export" the TX-pin of the ATMegas, so you don't have the chance to get any serial response.

Can you take a picture of the board an the way you connected the Pololu?

neromir, 2024-06-16:
Sure! Pictures attached. I'm connected to the ISP breakout pins on the board. I assumed those would include serial tx/rx pins, but I'm not very familiar with using an ISP programmer like this and that may be a bad assumption. 

[image]
[image]

I don't think the Pololu has a mode that needs to be switched, since I didn't need to do anything for my loopback test-- just jumper the tx and rx pins as seen [here](https://www.pololu.com/docs/0J67/3) and use the device's TTL port, rather than the device's programmer port on the USB connection to the computer (and looking at that page now makes me think that my assumption earlier was definitely bad).

Do you think I need to connect to the arduino board's pins directly (similar to how the FLTHY has a header for doing) or to the overall tx/rx pins like I have done for the ESP8266 experiment (see below)? Although for that I'd be worried about only being able to talk to the master.

[image]

RealNobser, 2024-06-16:
OK, you will not get any communication when just using the 6 pin ISP. The 6 pins are MISO, SCK, RESET, VCC, MOSI and GND. ISP does not include RX/TX!

There are 2 possibilities:
- If you want to communicate directly with one of the three Atmel boards, you have to add the headers on the left an connect your programmer to TXO and RXI.
- You connect your programmer to XBEE-RX. 

As I said above, the AIO1.5 has no pin for receiving messages from the Dome Master or Slave. So to get feedback from the MarcDuino/BetterDuino software you have to add the headers as described or you separate them from the AIO and communictate directly to them.

This ist not a BetterDuino Issue, it is by design of the hardware, unfortunately.

closed 2024-06-16 by RealNobser

neromir, 2024-06-16:
Sure enough, that was my issue. I rigged something up to connect to the TXO and RXI pins and it's working as expected now. Thanks!