RealNobser/BetterDuinoFirmwareV4 #11: Option to use different versions of R2D2`s Dome

Buddinski88, 2024-07-31:
### Situation
While learning the dome panels/servos, I noticed that the new MK4 dome is a problem. There are two new panels in the lower dome area and an ‘old’ one, which is known as S4 for the MK3 dome, has been removed.

### Complication
Due to this state, it is not clear which connections the servos should be connected to. In addition, sequences can no longer be displayed correctly.

### Solution
Creation of a wrapper that uses a different configuration for the assignment of the servo connections on the Benduino Dome Pro depending on the set version of the dome.
There should also be instructions for this in the documentation.

RealNobser, 2024-08-01:
I asked Michael Baddeley to add an animation option for the big panel, that is normaly atteched to Master Servo 6. He agreed. Please connect the new tiny panels to Slave Servo 10 and Slave Servo 11. I will adapt the animation sequences within the upcoming releases,

New Panel to Servo mapping here: https://github.com/RealNobser/BetterDuinoFirmwareV4?tab=readme-ov-file#connecting-servos-r2d2-common-way

Buddinski88, 2024-09-08:
I am closing the bug as the current implementation now allows to use the MK4 dome as well. Thank you 👍

closed 2024-09-08 by Buddinski88