RealNobser/BetterDuinoFirmwareV4 #14: Building for Chooper

darkoperator, 2025-05-09:
I'm trying to build dedicated hex files for chopper and to test DY player. I do not see in config the option for DEDICATED_FIRMWARE. What changes to the config.h should I make to achieve this

RealNobser, 2025-05-10:
Uh, there broke something on the latest commit. Will check ASAP.

RealNobser, 2025-05-10:
Use these settings for Chopper build with DY-Player option:

```cpp
// Specific Astromech, Choose one
// #define R2D2
#define CHOPPER
// #define BT1

// #define DEBUG_MSG                        // Additional Log Messages to Serial

#define INCLUDE_DOME_MASTER // Uncomment to include Dome Master Controller in combined firmware
#define INCLUDE_DOME_SLAVE	// Uncomment to include Dome Slave Controller in combined firmware
#define INCLUDE_BODY_MASTER // Uncomment to include Body Master Controller in combined firmware

// #define INCLUDE_CLASSIC_I2C_SUPPORT  // Uncomment to include classic Serial to I2C support (excludes body master built)
#define INCLUDE_I2C_SLAVE // Uncommend for new I2C slave mode
// #define INCLUDE_HOLO_RGB		// Uncomment for NeoPixel-Holo (excludes body master built)
#define ARDUINO_NEOPIXEL // Choose one!
// #define FASTLED_NEOPIXEL		// Choose one!
// #define SEPARATE_DOMELIFT    // uncomment, if you want to use (Master) AUX as separate Dome Lift Serial
#define INCLUDE_DY_PLAYER       // Uncomment to include DY sound module (not recommended at the moment, to many problems)
// #define INCLUDE_HOLO_TEST
```

darkoperator, 2025-05-10:
it worked, compiled firmware and was able to flash marcduino. Any way to make it a dedicated one (master, slave, body master) with debug?

RealNobser, 2025-05-11:
Sure. Uncomment DEBUG_MSG. Then choose just one option. INCLUDE_DOME_MASTER or INCLUDE_DOME_SLAVE or INCLUDE_BODY_MASTER.

```cpp
// Specific Astromech, Choose one
// #define R2D2
#define CHOPPER
// #define BT1

#define DEBUG_MSG                        // Additional Log Messages to Serial

#define INCLUDE_DOME_MASTER // Uncomment to include Dome Master Controller in combined firmware
// #define INCLUDE_DOME_SLAVE	// Uncomment to include Dome Slave Controller in combined firmware
// #define INCLUDE_BODY_MASTER // Uncomment to include Body Master Controller in combined firmware
```

closed 2025-05-11 by RealNobser